VfN-NRW/gluon-firmware-builder #6: Konfig mode startet nicht, LUA Fehlermelung

bitnukl, 2016-07-17:
/usr/lib/lua/luci/dispatcher.lua:364: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:364: Failed to execute function dispatcher target for entry '/gluon-config-mode'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:364: Failed to execute cbi dispatcher target for entry '/gluon-config-mode/wizard'.
The called action terminated with an exception:
.../lib/lua/luci/model/cbi/gluon-config-mode/wizard.lua:18: wrong number of arguments to 'insert'
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:364: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:135: in function </usr/lib/lua/luci/dispatcher.lua:134>

RubenKelevra, 2016-07-19:
Frage ob das in der letzten Version auch noch auftritt.

bitnukl, 2016-07-19:
Jap, jetzt auch auf v11

RubenKelevra, 2016-07-21:
Wir haben gestern vermutet das dieser Commit läuft: https://github.com/VfN-NRW/gluon-firmware-builder/commit/3ab4006328a9bacabbd12565bb17aa1b881252d5 

Nein tut er nicht.

RubenKelevra, 2016-07-21:
Ich hab https://github.com/VfN-NRW/gluon-firmware-builder/commit/d5e036c29bdec9bf741925ba3ceddf4690491e96 probiert, das funktioniert.

RubenKelevra, 2016-07-21:
Ich hab https://github.com/VfN-NRW/gluon-firmware-builder/commit/8a52ccb5938f280f1ab79f047eac3ed61df7dc96 probiert, funktioniert auch.

RubenKelevra, 2016-07-21:
https://github.com/VfN-NRW/gluon-firmware-builder/commit/eb879ca9e890e4248353fa0220555ee02c810d85 ist kaputt

RubenKelevra, 2016-07-21:
Das macht irgendwie keinen Sinn.

bitnukl, 2016-07-23:
fixed

closed 2016-07-23 by bitnukl