VfN-NRW/gluon-firmware-builder #53: LAN-WAN-Bridge Patch hinzufügen

RubenKelevra, 2016-07-20:


RubenKelevra, 2016-08-01:
https://github.com/VfN-NRW/gluon/commit/8fc6e8e13d2f06ed136267b0e2b9a2b5577a2180

https://github.com/VfN-NRW/gluon/commit/9a23013a3fb0890f9fb78a3960854e7e58255e27

https://github.com/VfN-NRW/gluon/commit/f7b466208bfc7ca43576cbe2b1e2949218357142

https://github.com/VfN-NRW/gluon/commit/96fb736daa67b46cb7d50290deb8bb92c24deb15

https://github.com/VfN-NRW/gluon/commit/54a8e56ada2625868b41c31447f0c91bfd6a56d1

https://github.com/VfN-NRW/gluon/commit/174e239038e6dfaf51961c096e486102a211d5f2

RubenKelevra, 2016-08-02:
https://github.com/VfN-NRW/gluon/commit/32a8d630a49c3626d5f8489c37ee7e30dee1a62e

RubenKelevra, 2016-08-02:
added: https://github.com/VfN-NRW/gluon-firmware-builder/commit/aa345260f8515180d3fdb097b1972b83e6b068f2

bitnukl, 2016-08-02:
```
#!/usr/bin/lua

local lutil = require 'luci.util'
local sysconfig = require 'gluon.sysconfig'
local uci = require('luci.model.uci').cursor()

for _, lanif in ipairs(lutil.split(sysconfig.lan_ifname, ' ')) do
    for clientif in uci:get('network', 'client', 'ifname') do
        if clientif == lanif then
            uci:section('gluon-node-info', 'void', 'bridge_lan', { enabled = 0, })                    
        end
    end
    for wanif in uci:get('network', 'wan', 'ifname') do
        if wanif == lanif then
            uci:section('gluon-node-info', 'void', 'bridge_lan', { enabled = 1, })                    
        end
    end
end
uci:save('gluon-node-info')

```

bitnukl, 2016-08-02:
```
if not uci:get('gluon-node-info', 'bridge_lan') then
 uci:section('gluon-node-info', 'void', 'bridge_lan', { enabled = 1, })
end
```

RubenKelevra, 2016-08-07:
https://github.com/VfN-NRW/gluon-firmware-builder/commit/a6248ecaee9f0da3896c0cddf47cf3341fe5419e

closed 2016-08-12 by RubenKelevra