VfN-NRW/gluon-firmware-builder #70: wlan config wird nicht richtig erstellt

bitnukl, 2016-07-22:
config wifi-device 'radio0'
        option type 'mac80211'
        option channel '11'
        option hwmode '11g'
        option path 'platform/qca953x_wmac'
        option htmode 'HT20'
        option noscan '1'
        option country 'DE'
        option supported_rates '6000 9000 12000 18000 24000 36000 48000 54000'
        option basic_rate '6000 9000 18000 36000 54000'
        option iapp_interface 'bat0'
        option max_inactivity '120'

RubenKelevra, 2016-07-22:
Wifi channel fix https://github.com/VfN-NRW/gluon-firmware-builder/commit/ecfcdb9bd662928a86e9dd8ff8e5fa8aadba0349

RubenKelevra, 2016-07-22:
HT-Mode fix https://github.com/VfN-NRW/gluon-firmware-builder/commit/d0081c066d3fe26c6ceacda32752f1ab04a48890

closed 2016-07-22 by RubenKelevra