VfN-NRW/gluon-firmware-builder #89: Der Autoupdater ist derzeit sehr ressourcenintensiv, analyse ob man das verbessern kann

RubenKelevra, 2016-07-25:
Derzeit lädt der Autoupdater offenbar alle 60 Minuten die etwa 45 KByte großes Manifest-Datei herunter, nur um zu prüfen ob die Version aktualisiert wurde, und wenn, ob das Datum sehr alt ist. Zwei keine Werte für die eine große Datei geladen wird. Plus HTTP Overhead kommt man dabei auf etwa 1,5 MB pro Router und Tag.

Das ganze System wirkt auf mich nicht wirklich optimiert, auf dem Update-Server wird so ein monatlicher Traffic von 50 GB generiert, bei 1100 Nodes, dafür das manchmal gar kein Update in einem Monat ausgeliefert wird.

Die Informationen sollten in einzelne kleinere Dateien ausgelagert werden, und nur wenn es einen Anlass gibt, sollte ein Laden der Manifest-Datei angestoßen werden.

RubenKelevra, 2016-07-25:
Frage ist zusätzlich ob er jede Stunde bei jedem Remote die Manifest-Datei ließt, das wäre dann aktuell:

24 (h) x 45 KByte x 8 (remotes) = 8,6 MB 

pro Router und Tag plus HTTP-Overhead

kevin-olbrich, 2016-07-25:
Im Manifest stehen alle Modelle oder? Kann das gerade nicht prüfen.

Falls ja:
Wenn wir für jedes Modell einen Feed machen, dann ist es signifikant kleiner und wir können gerätespezifisch updaten. Davon ab reicht dann ein Check alle 6 Stunden.

RubenKelevra, 2016-07-25:
Ja.

Eine Aufteilung ist nicht möglich. Ich sehe keinen Vorteil in gerätespezifischen Updates. Die Checks zu reduzieren haben wir auch schon im Auge, danke für den Vorschlag.