VfN-NRW/gluon-firmware-builder #104: Gummersbacher Site.conf wird defekt generiert

RubenKelevra, 2016-07-28:
```
make[4]: Entering directory '/home/ruben/build/gluon-firmware-builder_a2/gluon08/openwrt/tools/b43-tools'
make[4]: Leaving directory '/home/ruben/build/gluon-firmware-builder_a2/gluon08/openwrt/tools/b43-tools'
make[3]: Leaving directory '/home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt'
make[2]: Leaving directory '/home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt'
touch /home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/target-prepared
make[1]: Leaving directory '/home/ruben/build/gluon-firmware-builder_a2/gluon08/openwrt'
make[1]: Entering directory '/home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt'
rm -f /home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt/key-build /home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt/key-build.pub
cp /home/ruben/build/gluon-firmware-builder_a2/gluon08/build/gluon-opkg-key /home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt/key-build
cp /home/ruben/build/gluon-firmware-builder_a2/gluon08/build/gluon-opkg-key.pub /home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt/key-build.pub
make[1]: Leaving directory '/home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt'
make[1]: Entering directory '/home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt'
/home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt/staging_dir/host/bin/lua: ...-firmware-builder_a2/gluon08/scripts/site_config.lua:9: [string "site.conf"]:136: unexpected symbol near '}'
stack traceback:
        [C]: in function 'assert'
        ...-firmware-builder_a2/gluon08/scripts/site_config.lua:9: in main chunk
        [C]: ?
Your site configuration did not pass validation.
make[1]: *** [/home/ruben/build/gluon-firmware-builder_a2/gluon08/Makefile:347: prepare] Error 1
make[1]: Leaving directory '/home/ruben/build/gluon-firmware-builder_a2/gluon08/build/ar71xx-generic/openwrt'
```

bitnukl, 2016-07-28:
ich schau es mir mal an...

bitnukl, 2016-07-28:
ok du warst schneller. :-/

RubenKelevra, 2016-07-28:
Das Problem ist irgendwo in den Remote-Hosts nehm ich an, nur da haben wir so einen Blob von JSON-Syntax

bitnukl, 2016-07-28:
Baue das gerade lokal, auf den ersten blick würde ich sagen da sind zwei klammern zu viel am ende

RubenKelevra, 2016-07-28:
Dann hat Marienheide das selbe Problem, da sieht die Syntax genau gleich aus.

bitnukl, 2016-07-28:
Baue gerade mit zwei klammern weniger

RubenKelevra, 2016-07-28:
Push das mal, dann bau ich das im nächsten Zyklus mit den beiden Städten :)

bitnukl, 2016-07-28:
ist durchgelaufen, bitte testen

RubenKelevra, 2016-07-28:
Ist behoben :)

closed 2016-07-28 by RubenKelevra