VfN-NRW/gluon-firmware-builder #120: Jenkins: Automatische Tests

RubenKelevra, 2016-07-29:
Ich würde gern die Images für KVM bauen und dort booten und probieren, damit online zu kommen.

Durch eine statische MAC-Adresse könnte eine permanente Freischaltung passieren. Der automatisch Test könnte darauf warten, dass der Node bei der API als online gemeldet wird, eine IP via Radvd/DHCP probieren von der Maschine zu beziehen, den Node-DHCP-Server prüfen.

RubenKelevra, 2016-07-29:
Zusätzlich könnten wir automatisiert die Setup-Page ausfüllen und abschicken. Das verursacht häufiger Probleme wie wir festgestellt haben. :)

kevin-olbrich, 2016-08-07:
Dafür brauchen wir einen zweiten Host. Sowas macht man nicht auf dem selben Server auf dem auch Jenkins läuft. Stattdessen kannst du das aber auch in z.B. OpenStack booten und die Infos über die (Python-)API austauschen.

RubenKelevra, 2016-08-07:
Halte ich für überflüssig, danke trotzdem für deinen Hinweis.