VfN-NRW/gluon-firmware-builder #140: 5 GHz Channel überdenken

RubenKelevra, 2016-08-05:
@bitnukl schaut so als wäre Channel 108 ne blöde Wahl:
[image]

bitnukl, 2016-08-05:
Wenn 80er Kanäle willst, können wir auch auf 100 gehen, 108 und 112 ist nur am wenigsten los.
Wenn du 40 Kanäle machst sind da am wenigsten WLANs, kein Radar und keine 5GHz Videobrücken und so etwas.

RubenKelevra, 2016-08-05:
Die Geräte machen wenn sie können VHT80, entsprechend wäre wohl 100 besser. Wenn das Wetterradar vorbei kommt könnte es sonst sein dass 5 GHz ausfällt. Weiß nicht ob der Treiber das kann mit auf 40 MHz limitieren ...

RubenKelevra, 2016-08-05:
In jedem Fall sollten wir den Kanal benutzen der einen zweiten VHT80 daneben zulässt :)

bitnukl, 2016-08-06:
done!
Wir sind auf Kanal 100

closed 2016-08-06 by bitnukl