VfN-NRW/gluon-firmware-builder #150: [Backend?] No links drawn on map after update from 0.8

RubenKelevra, 2016-08-10:
Matthias discovered the following issue:

Ich habe meine Nodes (TL-WR841N) V10/V11 auf die FW 4.2a2 hochgerüstet. Vorher war die 0.8 drauf. Wenn ich auf die Karte schaue sehe ich das die Geräte sich untereinander nicht verbinden sondern lediglich zu einem Knoten im Nachbarhaus. Unter der 0.8 waren einige Verbindungen untereinander aufgebaut. Nur ein Anzeigefehler oder tatsächlich ein Problem?
Meine Knoten sind die crueptx-xx, Nachbar ist Burger-Linde02

http://map.vfn-nrw.de/?7.320894043907296-51.02006355853635-20-2282

RubenKelevra, 2016-08-10:
Rückfrage ob das Problem nach Update auf die 4.2.0 Beta1 imernoch auf tritt.

bitnukl, 2016-08-10:
War noch bevor wir das auf dem Server gefixt haben.

closed 2016-08-10 by bitnukl