VfN-NRW/gluon-firmware-builder #155: 0.8.0 hat kaputte update pfade

bitnukl, 2016-08-10:
beta: 
        list mirror 'http://[fddf:ebfd:a801:fff::1]/fff/remscheid/beta/'
        list mirror 'http://[fddf:ebfd:a801:fff::2]/fff/remscheid/beta/'
        list mirror 'http://[2001:67c:1714:fff:1]/fff/remscheid/beta/'
        list mirror 'http://[2001:67c:1714:fff:2]/fff/remscheid/beta/'
        list mirror 'http://[2001:4ba0:92f3:300::100]/fff/badbentheim/beta/'
        list mirror 'http://[2001:41d0:2:b0db:1::67]/fff/remscheid/beta/'
        list mirror 'http://62.141.34.119/fff/remscheid/beta/'
        list mirror 'http://5.135.22.67/fff/badbentheim/beta/'
        option good_signatures '1'
        option name 'rc'  
        list pubkey '1edde86c682e36260258c49ae55f0a6e07c058d6ac806a6d30a903ff6d4848d8'
        list pubkey '6cdb4e7eb5ddc4ba57d336dfc86b7fe7db647acc077407824030b6fc0f4f91bc'
        list pubkey 'b5270666ed74c74965ae9188d71282504c912ec19dccc6ead2f63ac2791e0c7f'
        list pubkey '33d1fe8cb1edd20995fdeb0d70559911d55ee07bad344e91c54567edf716c54d'

stable:
        list mirror 'http://[fddf:ebfd:a801:fff::1]/fff/remscheid/stable/'
        list mirror 'http://[fddf:ebfd:a801:fff::2]/fff/remscheid/stable/'
        list mirror 'http://[2001:67c:1714:fff:1]/fff/remscheid/stable/'
        list mirror 'http://[2001:67c:1714:fff:2]/fff/remscheid/stable/'
        list mirror 'http://[2001:4ba0:92f3:300::100]/fff/badbentheim/stable/'
        list mirror 'http://[2001:41d0:2:b0db:1::67]/fff/remscheid/stable/'
        list mirror 'http://62.141.34.119/fff/remscheid/stable/'
        list mirror 'http://5.135.22.67/fff/badbentheim/stable/'
        option good_signatures '2'
        option name 'stable'
        list pubkey '1edde86c682e36260258c49ae55f0a6e07c058d6ac806a6d30a903ff6d4848d8'
        list pubkey '6cdb4e7eb5ddc4ba57d336dfc86b7fe7db647acc077407824030b6fc0f4f91bc'
        list pubkey 'b5270666ed74c74965ae9188d71282504c912ec19dccc6ead2f63ac2791e0c7f'
        list pubkey '33d1fe8cb1edd20995fdeb0d70559911d55ee07bad344e91c54567edf716c54d'

RubenKelevra, 2016-08-11:
Das ist kein Problem, wir würden einfach die ohne badbentheim benutzen. :)

closed 2016-08-11 by RubenKelevra

bitnukl, 2016-08-11:
Und damit wir das nicht vergessen, habe  ich es hier eingetragen, du Nuss! :-D

RubenKelevra, 2016-08-11:
Nein, ich meinte damit, ich hab mir schon eine Update-IP ausgesucht und konfiguriert, diese wird von diesem Bug nicht tangiert. Für die neue Version ist es gefixt, also betrifft uns das nicht mehr. :)