VfN-NRW/gluon-firmware-builder #159: Analysieren warum der Durchsatz bei Überlast so stark einbricht

RubenKelevra, 2016-08-13:
@bitnukl und ich haben gestern in einem kleinen Testaufbau mit Traffic-Generatoren einen Durchsatztest mit L2 Paketen gemacht. Hier ist aufgefallen, dass wenn durch eine Adhoc Strecke über ein Batman-Netz zu viele Daten geschickt werden, der Durchsatz überdurchschnittlich einbricht.

Beispielsweise der Kanal schafft 120 Mbit/s hin und 1 Mbit/s wird zurück geschickt.

Wenn nun 150 MBit/s reingeschickt werden, kommen lediglich 3 Mbit/s an.

Werden 600 MBit/s reingeschickt kommen auch 3 Mbit/s an, was darauf schließen lässt, dass es sich hier nicht um eine CPU-Überlastung handelt sondern eine schlechte Behandlung von WLAN-Hardware.

kevin-olbrich, 2016-08-13:
@RubenKelevra willst du das nicht lieber mal upstream geben und zusammen analysieren?

RubenKelevra, 2016-08-13:
Nein