VfN-NRW/gluon-firmware-builder #162: Falsche TX-Power bei 4.2

RubenKelevra, 2016-08-15:
```
# iwinfo
client0   ESSID: "wk.freifunk.net"
          Access Point: C2:4C:01:2D:31:4A
          Mode: Master  Channel: 5 (2.432 GHz)
          Tx-Power: 14 dBm  Link Quality: 32/70
          Signal: -78 dBm  Noise: -95 dBm
          Bit Rate: 22.6 MBit/s
          Encryption: none
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0

ibss0     ESSID: "mesh-v15"
          Access Point: 02:CA:FE:15:21:96
          Mode: Ad-Hoc  Channel: 5 (2.432 GHz)
          Tx-Power: 14 dBm  Link Quality: 29/70
          Signal: -81 dBm  Noise: -95 dBm
          Bit Rate: 10.7 MBit/s
          Encryption: unknown
          Type: nl80211  HW Mode(s): 802.11bgn
          Hardware: unknown [Generic MAC80211]
          TX power offset: unknown
          Frequency offset: unknown
          Supports VAPs: yes  PHY name: phy0
```

bitnukl, 2016-08-15:
Bei mir macht..
841:
Tx-Power: 16 dBm
740N:
Tx-Power: 13 dBm

RubenKelevra, 2016-08-15:
Jup das passt

RubenKelevra, 2016-08-15:
Denke das ist einfach die Limitierung bei den größeren vom Chip her. :)

closed 2016-08-15 by RubenKelevra