VfN-NRW/gluon-firmware-builder #165: Pro/Contra Wechsel auf WireGuard+Babel

RubenKelevra, 2016-08-17:
Ich weiß, das ist viel Umbau, aber würde gern mal prüfen wie unsere Optionen sind:

Contra:
-  Babel hat von der aktuellen Version zur dev-version breaking changes

Pro:
- Kann HMAC-Authentifizierung von Nodes
- Füllt nur Kernel-Tabellen, entsprechend wird das Routing im Kernelspace durchgeführt
- Encryption passiert im Kernelspace, entsprechend schnell ist das auf den Nodes
- Reduziert den Overhead extrem auf den Gateways durch Kernelspace Implementierung
- Leichte Erhöhung der Sicherheit da WireGuard eine Crypto-Generation weiter ist als Fastd

Zu Prüfen:

- [ ] Geht Roaming kaputt?
- [ ] Geht Multicast-Routing?
- [ ] Wie skaliert das?
- [ ] Unbekannt wie schnell WireGuard auf den Nodes läuft
- [ ] Wie skaliert WireGuard auf dem Gateway, erzeugt das auch ein Interface pro Peer?
- [ ] Wie gut ist Babeld debuggbar wenn was schief läuft, lässt das auch gut Metriken exportieren?
- [ ] Wie aktiv wird Babeld weiterentwickelt?
- [ ] Wieviel Aufwand ist die Implementierung in Gluon (wie weit ist der Fortschritt?) ?

kaechele, 2016-09-05:
Also. Gerade mal getestet. iperf3 über Wireguard zwischen Router (Archer C7 v2 mit aktuellstem LEDE) und Fedora 24 VM bei OVH macht netto ziemlich genau 40 Mbit/s down und 9 Mbit/s up (ohne Wireguard sind 43,5/9,33 Mbit/s gemessen worden). Max. MSS war dabei 1368 (macht 1408 payload MTU).

Es gibt für alle Peers nur ein Interface.
Allerdings findet keine Layer2 Vollvermaschung statt, sondern Routing (Layer3).

Beispielaufbau:
`A <-----> B <------> C`
A und C können B direkt erreichen (traceroute 1 hop)
A kann C über B erreichen (traceroute 2 hops, erster hop ist B)
Das gleiche gilt für C nach A. Geht auch über B.

RubenKelevra, 2016-09-05:
Danke für deine Arbeit :)

RubenKelevra, 2016-09-05:
Ist damit die Geschwindigkeit auf den Nodes auch geklärt?

closed 2017-08-15 by RubenKelevra